waldoj/home-battery #9: Get a PowerWall-like device?

waldoj, 2019-03-11:
There are a bunch of companies that make PowerWall-like things, e.g. [SMA's Sunny Boy Storage](https://www.sma-america.com/products/battery-inverters/sunny-boy-storage-38-us-50-us-60-us.html), which [goes for $2,300](https://www.wholesalesolar.com/2931750/sma/inverters/sma-sunny-boy-storage-sbs-3.8-us-10-inverter), though that doesn't even include batteries. Those might be worth considering.

waldoj, 2019-03-11:
The LG Chem RESU looks good for this, too.

waldoj, 2019-03-11:
These all assume the existence of an inverter (per #1).

jathayde, 2019-03-11:
Did you check out the Nickel Iron batteries? https://ironedison.com

waldoj, 2019-03-11:
I just added those as a candidate over on #6 — thanks!